TinkerBoard/debian_kernel #2: System "reboot" issue by using "sysrq"

sghazagh, 2017-05-28:
I have an "init" file for resizing the rootfs partition in first boot.
I need to use magic SysRq key "**`echo b > /proc/sysrq-trigger`**" to reboot the system after partition has resized.

The "sysrq" options in Kernel are enabled however, system cannot reboot.
It shut-downs the system immediately as expected but doesn't initiate the boot procedure.
Have to manually power-off and power-on the power source to make it boot again.

Can you please investigate and provide a solution?!
Thanks

tido-, 2017-06-02:
for kernel 4.4 https://forum.armbian.com/index.php?/topic/3327-asus-tinkerboard/&do=findComment&comment=31067

sghazagh, 2017-06-02:
Hi Tido,
Thanks for the reply.
But I downloaded the kernel after 5th/May so it should already have been fixed but it's not.
Can you please give me the commit or link showing the changes then I check and see if patched has applied?

I do appreciate your help.
(you always are helpful like what you did in Lemaker Forums :)

tido-, 2017-06-03:
Hi sghazagh,

I also thought I have seen name before - same to you, thank you.
I was not compiled with the fixes until now, see here: 
https://forum.armbian.com/index.php?/topic/3327-asus-tinkerboard/&do=findComment&comment=32887

TinkerTeam, 2017-06-14:
Hi @sghazagh & @tido- ,

Thanks for your reply.
And it will be fixed in the next release.

TinkerTeam, 2017-07-21:
Fix on 15a531028f80eef8f2385173ec39d370cac88486.

closed 2017-07-21 by TinkerTeam