TinkerBoard/debian_kernel #3: Interrupt load with full speed USB devices

MusicMiK, 2017-06-26:
I noticed an odd behavior with full speed USB devices and had a closer look, with the following results:

- Nothing connected to USB: about 200 interrupts/s (checked with vmstat)
- Wireless keyboard/mouse receiver connected (USB HID device): >8000 interrupts/s
- USB MIDI device (E-Mu Xmidi 2x2 or M-Audio Keystation 32, same behavior), using amidi -d to just dump the messages: >120000 interrupts/s

High speed USB devices don't show a behavior like this.

scorpiochang, 2017-08-17:
Hi Mik
The USB IP of Tinker board is DWC2, it need dwc_otg driver handle more interrupt about SOF/EOF,....
There is an patch can reduce the unused SSPLIT/CSPLIT NAK interrupts and down to 24k/s after you execute "cat /dev/ttyUSB0".
Have a nice day.
[0001-USB-fix-handle-NAK-for-IN-OUT-SSPLIT-CSPLIT-transfer.zip](https://github.com/TinkerBoard/debian_kernel/files/1230473/0001-USB-fix-handle-NAK-for-IN-OUT-SSPLIT-CSPLIT-transfer.zip)

MusicMiK, 2017-08-17:
Hi,

thanks. I meanwhile switched to a different SoC for my project(s), my TinkerBoard is now just a media player. But i'll have a look at the patch when i feel the need to tinker with it once again.

...Michael

closed 2018-03-14 by jamess-huang