TinkerBoard/debian_kernel #4: use device tree for RPi 7" touchscreen/MCU

Tonymac32, 2017-11-11:
Instead of sticking it in rockchip.c

I've already done this on Armbian Legacy kernel, leaving the drivers unchanged, leaving the I2C entries out of rockchip.c.
added to rk3288-miniarm.dts:
```
&i2c3 {
status = "okay";

tinker_ft5406:tinker_ft5406@38 {
    compatible = "tinker_ft5406";
    reg = <0x38>;
};

tinker_mcu:tinker_mcu@45 {
    compatible = "tinker_mcu";
    reg = <0x45>;
};

};
```

Also, the tinker_mcu and asus_mcu are redundant but different (unless there is some special hardware floating around?) I'm using the tinker_mcu driver successfully, omitted the asus_mcu files entirely. Just thought one less bit of Tinker-specific kernel code to keep track of might be nice.

jamess-huang, 2017-11-13:
Hi, Tony
   We think your comment is good. Can you give us the patch or we can modify this by ourselves ?

Tonymac32, 2017-11-14:
Hello James,

https://github.com/TinkerBoard/debian_kernel/pull/5

This is, I believe complete, as always test.

Another bit I had to do to keep the Tinker board and other RK3288 boards on a common kernel is to check the board DT compatible on the reboot workarounds.
https://github.com/armbian/build/blob/master/patch/kernel/rockchip-default/135_tinker_boot_fix.patch

jamess-huang, 2017-11-15:
Hi, Tony
    Thanks for the reminder, we will follow the armbian commit to add the DT compatible check.

jamess-huang, 2018-01-02:
already added into our code base, close it

closed 2018-01-02 by jamess-huang