TinkerBoard/debian_kernel #6: add JustBoom audio boards?

shawaj, 2017-12-01:
Hi,

We would like to add the JustBoom audio boards to your platform. What is the process for this?

Thanks

jamess-huang, 2017-12-04:
We have not yet tried the JustBoom board but we are glad to try this board for you.
Can you give us the model name which you want to use ? We will try to get the same one and check the function.

By the way, we can support the Hi-Fi berry and IQAudio DAC boards.

shawaj, 2017-12-04:
I think you just replied to my email also :-)

I can send you free samples of all of our boards to add support!

Thanks for your reply

shawaj, 2017-12-04:
Sending the info from email here too, for reference:

About guides - we have a lot of guides here:

https://www.justboom.co/technical-guides/
https://www.justboom.co/start/
https://www.justboom.co/software/

We also have GitHub here:
https://github.com/PiSupply/JustBoom

We also have our drivers already in the Raspbian Kernel here:
https://github.com/raspberrypi/linux/blob/204050d0eafb565b68abf512710036c10ef1bd23/sound/soc/bcm/justboom-dac.c
https://github.com/raspberrypi/linux/blob/204050d0eafb565b68abf512710036c10ef1bd23/sound/soc/bcm/justboom-digi.c

And device tree overlays:
https://github.com/raspberrypi/linux/blob/204050d0eafb565b68abf512710036c10ef1bd23/arch/arm/boot/dts/overlays/justboom-dac-overlay.dts
https://github.com/raspberrypi/linux/blob/204050d0eafb565b68abf512710036c10ef1bd23/arch/arm/boot/dts/overlays/justboom-digi-overlay.dts

I hope that helps - if you have any further questions, please ask.

Also - we have other products such as gesture sensing, ePaper displays, UPS / battery power etc for Raspberry Pi - do you think we can support all of these? I can send samples of all products if that helps? We are interested to build a long term partnership :-)

jamess-huang, 2017-12-04:
Hi,
   Let's discuss in the email, I will close this issue. Thanks for your interest in Tinker Board.

closed 2017-12-04 by jamess-huang

shawaj, 2017-12-05:
@jamess-huang did you see my email yet?