TinkerBoard/debian_kernel #7: iotop doesn't work

olka, 2017-12-30:
> sudo iotop
Could not run iotop as some of the requirements are not met:
- Linux >= 2.6.20 with
  - I/O accounting support (CONFIG_TASKSTATS, CONFIG_TASK_DELAY_ACCT, CONFIG_TASK_IO_ACCOUNTING)

OS version: Linux tinkerboard 4.4.71+ #1 SMP Wed Nov 15 14:31:48 CST 2017 armv7l GNU/Linux

jamess-huang, 2018-01-02:
Thanks for the information , we will check it

jamess-huang, 2018-01-02:
After enable some configs, it can run iotop successfully. We will merged it into our code base.

diff --git a/arch/arm/configs/miniarm-rk3288_defconfig b/arch/arm/configs/miniarm-rk3288_defconfig
index 5eb17847..60aa685 100755
--- a/arch/arm/configs/miniarm-rk3288_defconfig
+++ b/arch/arm/configs/miniarm-rk3288_defconfig
@@ -818,3 +818,7 @@
 CONFIG_LIRC_STAGING=y
 CONFIG_STAGING_MEDIA=y
 CONFIG_I2S_SHORT=y
+CONFIG_TASKSTATS=y
+CONFIG_TASK_DELAY_ACCT=y
+CONFIG_TASK_XACCT=y
+CONFIG_TASK_IO_ACCOUNTING=y

closed 2018-02-06 by jamess-huang