TinkerBoard/debian_kernel #10: 4k playback issue

kundaria, 2018-03-19:
media player crash when i try to play H.264 or H.265 30fps videos

jamess-huang, 2018-03-20:
Hi, 
    May I know your image version ? and what command or application do you use to play the videos.
    It will help us to check, if you can provide the detail step,

kundaria, 2018-03-21:
i am trying to play

FightingForFun, 2018-03-21:
Mali Drivers on 2.0.5 not added ? 
https://developer.arm.com/products/software/mali-drivers/midgard-kernel

jamess-huang, 2018-03-22:
May I know what application do you use to play the video? or if you play the video over the terminal, please give us the command, thanks.

kundaria, 2018-03-30:
i am using default media player which comes preinstalled with tinker os 2.0.4

jamess-huang, 2018-04-02:
Hi
We will try the v2.0.4 image, in the meantime, could you help to test the v2.0.5 image ?
http://dlcdnet.asus.com/pub/ASUS/mb/Linux/TInker_Board/2GB/20180222-tinker-board-linaro-stretch-alip-v2.0.5.img.zip?_ga=2.96681752.687927311.1522636813-81248712.1509000501

jamess-huang, 2018-04-02:
Hi
    We have tried many videos which have the similar SPEC of you provided with image v2.0.4.
    Unfortunately, we cannot duplicate the same issue with you.
    Can you just provide your test video to us?

    NOTE : The videos we tried cab be downloaded at the below link
    http://jell.yfish.us/