TinkerBoard/debian_kernel #11: How to update the GPU

FightingForFun, 2018-03-21:
Good afternoon. Tell me how to update the driver on the GPU?

In the new version of the 2.0.5 system, the Midgard driver is installed - r9p0-05rel0 (Released Dec 13, 2016).

On the ARM.com website (https://developer.arm.com/products/software/mali-drivers / midgard-kernel) the Midgard driver is presented - r23p0-01rel0 (Released 9th March 2018).

How can I install the latest version of the drivers on the GPU with arm.com? Is there an instruction manual?

Thank you in advance.

FightingForFun, 2018-03-23:
And what about this ? https://github.com/rockchip-linux/rk-rootfs-build/tree/master/packages/armhf
Maybe I'm stupid, and it's not possible to update the drivers, so say it.

alex3788, 2018-03-28:
Hi FightingForFun,

/usr/lib/arm-linux-gnueabihf/libmali.so is symbolic link to /usr/lib/arm-linux-gnueabihf/libmali-midgard-r9p0-r0p0.so

It is not mali driver and the libmali-midgard-r9p0-r0p0.so is user space library that application can use it to communicate to mali driver

You can get update from [https://github.com/rockchip-linux/rk-rootfs-build/tree/master/packages/armhf/libmali](https://github.com/rockchip-linux/rk-rootfs-build/tree/master/packages/armhf/libmali) or [https://github.com/rockchip-linux/libmali](https://github.com/rockchip-linux/libmali)

FightingForFun, 2018-03-29:
After installing the drivers, stopped working Hardware acceleration in Chrome. Uses only CPU. Some programs have written (you using the 10.6 user-space mali drivers (and kernel is 10.4) ... I do not remember literally. And this programs do not want to run.

But at the same time, the number of FPS in GlxGears doubled!

Prompt, how it is possible to cure it?

alex3788, 2018-03-29:
Hi FightingForFun,

You can recovery by the below step:
Example:
1. dpkg-deb -I libmali-rk-midgard-t76x-r14p0-r0p0_1.6-1_armhf.deb       (check the installed package name)

You will see the information for the package

 new debian package, version 2.0.
 size 7276504 bytes: control archive=703 bytes.
      29 bytes,     1 lines      conffiles
     607 bytes,    14 lines      control
     269 bytes,     3 lines      md5sums
 **Package: libmali-rk-midgard-t76x-r14p0-r0p0**
 Source: libmali-rk
 Version: 1.6-1
 Architecture: armhf
 Maintainer: Randy Li <randy.li@rock-chips.com>
 Installed-Size: 17199
 Depends: libc6 (>= 2.15), libdrm2 (>= 2.3.1), libgcc1 (>= 1:3.5), libstdc++6 (>= 4.8), libx11-6, libx11-xcb1, libxcb-dri2-0 (>= 1.8), libxcb1
 Replaces: libegl1-mesa, libegl1-x11, libgbm1, libgles2-mesa
 Provides: libegl1, libegl1-mesa, libgbm1, libgles2-mesa, libwayland-egl1-mesa
 Section: libs
 Priority: optional
 Multi-Arch: same
 Homepage: http://www.rock-chips.com/
 Description: The mali library for Rockchip RK3288(not included RK3288W).

2, sudo dpkg -r libmali-rk-midgard-t76x-r14p0-r0p0              (this is remove package)
3. sudo dpkg  -i libmali-rk-midgard-r9p0-r0p0_1.5-4_armhf.deb    (reinstall r9p0 mali library)

FightingForFun, 2018-03-29:
I reinstall the system and try to install r14p0-r0p0:

```
root@tinkerboard:/home/linaro/Downloads# dpkg -i libmali-rk-midgard-t76x-r14p0-r0p0_1.6-1_armhf.deb
Selecting previously unselected package libmali-rk-midgard-t76x-r14p0-r0p0:armhf.
(Reading database ... 89266 files and directories currently installed.)
Preparing to unpack libmali-rk-midgard-t76x-r14p0-r0p0_1.6-1_armhf.deb ...
Unpacking libmali-rk-midgard-t76x-r14p0-r0p0:armhf (1.6-1) ...
dpkg: error processing archive libmali-rk-midgard-t76x-r14p0-r0p0_1.6-1_armhf.deb (--install):
 trying to overwrite '/usr/lib/arm-linux-gnueabihf/libEGL.so.1', which is also in package libmali-rk-midgard-r9p0-r0p0:armhf 1.5-4
Errors were encountered while processing:
 libmali-rk-midgard-t76x-r14p0-r0p0_1.6-1_armhf.deb
```
Ok. Next - > Complete remove via Synaptic Manager - libmali-rk-dev libmali-r9p0. And again try install 

```
root@tinkerboard:/home/linaro/Downloads# dpkg -i libmali-rk-midgard-t76x-r14p0-r0p0_1.6-1_armhf.deb
(Reading database ... 89461 files and directories currently installed.)
Preparing to unpack libmali-rk-midgard-t76x-r14p0-r0p0_1.6-1_armhf.deb ...
Unpacking libmali-rk-midgard-t76x-r14p0-r0p0:armhf (1.6-1) ...
Setting up libmali-rk-midgard-t76x-r14p0-r0p0:armhf (1.6-1) ...
```
Reinstall chromium. Check chrome://gpu

```
Graphics Feature Status
Canvas: Hardware accelerated
CheckerImaging: Disabled
Flash: Hardware accelerated
Flash Stage3D: Hardware accelerated
Flash Stage3D Baseline profile: Hardware accelerated
Compositing: Hardware accelerated
Multiple Raster Threads: Enabled
Native GpuMemoryBuffers: Software only. Hardware acceleration disabled
Rasterization: Hardware accelerated
Video Decode: Hardware accelerated
Video Encode: Hardware accelerated
WebGL: Hardware accelerated
WebGL2: Hardware accelerated
```

```
GL_VENDOR | VMware, Inc.
GL_RENDERER | Gallium 0.4 on llvmpipe (LLVM 3.9, 128 bits)
GL_VERSION | 3.0 Mesa 13.0.6
```

Before GPU driver update, GL_RENDERER: ibmali-midgard-r9p0-r0p0 all work it's normal.
After update Chromium start to work slowly. Big consume of CPU. Why now use Gallium? but not libmali-midgard-r14p0 ?

alex3788, 2018-03-30:
Dear FightingForFun,

Another way you can easily to update mali so version by `cp <source mali so> <destination mali so>`

You can get the mali so from here
[https://github.com/rockchip-linux/libmali/tree/rockchip/lib/arm-linux-gnueabihf](https://github.com/rockchip-linux/libmali/tree/rockchip/lib/arm-linux-gnueabihf)

For example:
sudo cp libmali-midgard-t76x-r13p0-r0p0.so /usr/lib/arm-linux-gnueabihf/libmali-midgard-r9p0-r0p0.so

This is my information:
GL_VENDOR | ARM
GL_RENDERER | Mali-T760
GL_VERSION | OpenGL ES 3.2 v1.r13p0-00rel0-git(a4271c9).40dbad8455f8b43d1f8fbb5a1fe733e6

FightingForFun, 2018-03-30:
Hi again :)

I download *.so -> execute command. After watch screen off - screen on. 

Check chrome://gpu - Software only on all. 
Reinstall chromium. Check.
Hardware acceleration but again use: Gallium 0.4. 

On glxinfo and screenfetch too Gallium 0.4. 

Argh... close this topic. because that no possible ))) Big thanks for answers!

alex3788, 2018-11-05:
You can get libmali-midgard-t76x-r13p0-r0p0.deb from here:
[https://drive.google.com/open?id=1KjO3kW8C1jaE56dGqnHmQXHr4jsBMcov](url)

closed 2019-01-29 by jamess-huang