TinkerBoard/debian_kernel #12: File not found: /lib/modules/4.4.71+/build/.config

FightingForFun, 2018-04-03:
Good day.

I have a problem with installing media_build (from LinuxTV for Mygica T230 usb TV adapter):
```
make[2]: Leaving directory '/home/linaro/media_build/linux'
./scripts/make_kconfig.pl /lib/modules/4.4.71+/build /lib/modules/4.4.71+/build 1
Preparing to compile for kernel version 4.4.71
File not found: /lib/modules/4.4.71+/build/.config at ./scripts/make_kconfig.pl line 33, <IN> line 4.
Makefile:377: recipe for target 'allyesconfig' failed
make[1]: *** [allyesconfig] Error 2
make[1]: Leaving directory '/home/linaro/media_build/v4l'
Makefile:26: recipe for target 'allyesconfig' failed
make: *** [allyesconfig] Error 2
can't select all drivers at ./build line 525

```
Linux-headers for 4.4.71 i not found. 
Need help.

FightingForFun, 2018-04-04:
Or how i can install media_build or put inside in Kernel?

jamess-huang, 2018-04-09:
You can git clone the kernel source code
$ git clone https://github.com/TinkerBoard/debian_kernel.git
and then create a soft link to the source code which you just download.
$ ln -s ~/debian_kernel /usr/src/linux-headers-$(uname -r)

closed 2018-07-25 by jamess-huang