TinkerBoard/debian_kernel #16: RTL8723BS P2P

jackmitch, 2018-07-09:
When creating a wifi-direct (p2p) connection with the new realtek driver the chip doesn't broadcast that it's available. The legacy access point appears, but the device can't be seen when searching from an Android phone, or by using p2p_find in wpa_supplicant from another p2p compatible device. This worked with the previous driver.

jamess-huang, 2018-07-10:
Dear, jackmitch
   May I know which branch do you use or which image version do you use ?
   If you compile the kernel by yourself, can you give me the wifi driver version ?
   You can check the drivers/net/wireless/rockchip_wlan/rtl8723bs/include/rtw_version.h for the version.

Sam2Hsu, 2018-07-16:
Hi jackmitch,

The rtl8723bs version on develop branch should be v5.1.7.4_24878.20171031_BTCOEX20170728-6b4e.
[rtw_version](https://github.com/TinkerBoard/debian_kernel/blob/develop/drivers/net/wireless/rockchip_wlan/rtl8723bs/include/rtw_version.h)

Could you help to check the version for each work and fail case.
Please check by below command.
`cat /proc/net/rtl8723bs/drv_cfg`

The result will look like
```
Kernel Version: 4.4.132+
Driver Version: v5.1.7.4_24878.20171031_BTCOEX20170728-6b4e
------------------------------------------------
CFG80211
RTW_USE_CFG80211_STA_EVENT
DBG:1
CONFIG_RTW_DEBUG
CONFIG_POWER_SAVING
LOAD_PHY_PARA_FROM_FILE - REALTEK_CONFIG_PATH=
RTW_DEF_MODULE_REGULATORY_CERT=0x00
CONFIG_TXPWR_BY_RATE_EN=0
CONFIG_TXPWR_LIMIT_EN=0
CONFIG_RTW_ADAPTIVITY_EN = 0
CONFIG_TX_AGGREGATION
MAX_XMITBUF_SZ = 20480
MAX_RECVBUF_SZ = 16256
```

Thanks,
Sam

jackmitch, 2018-07-17:
**Non-working:**

linaro@tinkerboard:~$ cat /proc/net/rtl8723bs/drv_cfg 

Kernel Version: 4.4.103+
Driver Version: v5.1.7.4_24878.20171031_BTCOEX20170728-6b4e
------------------------------------------------
CFG80211
RTW_USE_CFG80211_STA_EVENT
DBG:0
CONFIG_POWER_SAVING
LOAD_PHY_PARA_FROM_FILE - REALTEK_CONFIG_PATH=
RTW_DEF_MODULE_REGULATORY_CERT=0x00
CONFIG_TXPWR_BY_RATE_EN=0
CONFIG_TXPWR_LIMIT_EN=0
CONFIG_RTW_ADAPTIVITY_EN = 0
CONFIG_TX_AGGREGATION
MAX_XMITBUF_SZ = 20480
MAX_RECVBUF_SZ = 16256

**Working:**

linaro@tinkerboard:~$ cat /proc/net/rtl8723bs/drv_cfg 

Kernel Version: 4.4.71+
Driver Version: v4.4.0_17166.20160322_BTCOEX20151223-654a
------------------------------------------------
CFG80211
RTW_USE_CFG80211_STA_EVENT
DBG:0
CONFIG_POWER_SAVING
LOAD_PHY_PARA_FROM_FILE - REALTEK_CONFIG_PATH=
CONFIG_RTW_ADAPTIVITY_EN = 0
CONFIG_TX_AGGREGATION
MAX_XMITBUF_SZ = 20480
MAX_RECVBUF_SZ = 16256

Sam2Hsu, 2018-07-18:
Hi jackmitch,

We have confirmed this issue. P2P IE is not included in beacon frame.
If there are any updates, we'll let you know.

Thanks,
Sam

jackmitch, 2018-07-31:
Hi,

Any news on this? Have we asked Realtek to look into the issue so we're just waiting on them?

Regards,
Jack.

Sam2Hsu, 2018-08-01:
Hi jackmitch,

We are working with Realtek, it may take some time.
Sorry for your inconvenience.

Thanks,
Sam

Sam2Hsu, 2018-08-03:
Hi jackmitch,

Could you try below patch? It works for me.
After comment out below codes, driver will build a common P2P IE for Wi-Fi direct.

Previous driver also used this P2P IE, but which wasn't the real P2P IE got from wpa_supplicant.
It is just a workaround not the real solution for this issue.
```
diff --git a/drivers/net/wireless/rockchip_wlan/rtl8723bs/core/rtw_mlme_ext.c b/drivers/net/wireless/rockchip_wlan/rtl8723bs/core/rtw_mlme_ext.c
index 84266e399996..25b5b8848235 100755
--- a/drivers/net/wireless/rockchip_wlan/rtl8723bs/core/rtw_mlme_ext.c
+++ b/drivers/net/wireless/rockchip_wlan/rtl8723bs/core/rtw_mlme_ext.c
@@ -8062,7 +8062,7 @@ void issue_probersp(_adapter *padapter, unsigned char *da, u8 is_valid_p2p_probe
 	    /* IOT issue, When wifi_spec is not set, send probe_resp with P2P IE even if probe_req has no P2P IE */
 	    && (is_valid_p2p_probereq || !padapter->registrypriv.wifi_spec)) {
 		u32 len;
-#ifdef CONFIG_IOCTL_CFG80211
+#if 0
 		if (adapter_wdev_data(padapter)->p2p_enabled && pwdinfo->driver_interface == DRIVER_CFG80211) {
 			/* if pwdinfo->role == P2P_ROLE_DEVICE will call issue_probersp_p2p() */
 			len = pmlmepriv->p2p_go_probe_resp_ie_len;
```

jackmitch, 2018-08-04:
Hi Sam,

Thank you, this workaround has gotten me a working p2p setup for now. However this has also exposed that concurrent mode is now also broken in the new drivers. I'll keep digging into trying to fix concurrent mode and many thanks for looking into the p2p issue for me.

Kind Regards,
Jack.

Sam2Hsu, 2018-08-06:
Hi Jack,

If you need wlan and p2p at the same time. Try to add an EXTRA_CFLAGS in Makefile, it will generate two  interfaces wlan0 and p2p0.
`EXTRA_CFLAGS += -DCONFIG_CONCURRENT_MODE`

Hope this information could help you.

Thanks,
Sam

Sam2Hsu, 2018-08-15:
Hi Jack,

I tried to bring up wlan0/wlan1 manually, wlan0 for STA and wlan1 for P2P. Both interfaces were worked successfully.

Before bring up interface, I killed all existent wpa_supplicant processes.

Here's my supplicant configuration:

STA
```
ctrl_interface=/var/run/wpa_supplicant
update_config=1
country=US
``` 

P2P
```
ctrl_interface=/var/run/wpa_supplicant
update_config=1
device_name=tinkerboard
device_type=1-0050F204-1
p2p_go_intent=15
p2p_go_ht40=1
```

Thanks,
Sam

jackmitch, 2018-08-21:
Hi Sam,

Thanks for the update, turns out I wasn't using the same channel for each interface which was causing issues. Thanks for the follow up.

jamess-huang, 2018-11-21:
The solution has been submitted.
https://github.com/TinkerBoard/debian_kernel/commit/4f863e61c099651fbc9e48ae00925c05e219a61b

closed 2018-11-21 by jamess-huang

Azq2, 2022-11-27:
Possible fix for similar driver: https://github.com/jwrdegoede/rtl8189ES_linux/pull/87

But you need backport this changes to this driver