TinkerBoard/debian_kernel #17: How to upgrade?

p1r473, 2018-07-13:
I have read all of posts about upgrading via apt-get dist-upgrade or apt-get upgrade

I am still so confused after reading all posts...

I have made many changes and am running webserver etc on my tinkerboard so I do not want to reinstall

so.... can i upgrade to latest OS by  apt-get dist-upgrade or apt-get upgrade?

I understand it is not best approach.
But, will it work?

aBradleyUno, 2018-07-18:
If what you are trying to upgrade is the kernel then you would have to compile it and place the binary image in your boot partition, otherwise if what you are trying to do is upgrade your distro then try this. https://linuxconfig.org/how-to-upgrade-debian-9-stretch-to-debian-10-buster

jamess-huang, 2018-07-19:
Dear, p1r473
We are sorry that we now have no method to update rootfs from apt. But we have the plan to update the kernel from apt and already verified pass at internal.
If you need, I can give you the kernel package to update.

p1r473, 2018-07-19:
Do you have any estimated timelinefor when I can update rootfts from apt?

plchampigny, 2018-09-04:
@jamess-huang I would like to have someway to upgrade my current build from 2.0.5 to 2.0.7. We configured an image with that version and it would be painful to recreate the image. We need that version upgrade because a hdmi issue has been fixed in that version.

jamess-huang, 2018-09-05:
Hi, plchampigny
    As posted, we are sorry that we have no method that can update "full" image from one version to another. But we can only update the kernel image by replace the zImage and rk3288-miniarm.dtb in the sdcard. Maybe you can try to only update kernel image to check the hdmi issue. If ok, I will upload the kernel image of v2.0.7.

plchampigny, 2018-09-19:
Ok so how can I do that? Is there something I can follow to proceed with the kernel update?

jamess-huang, 2018-09-20:
https://drive.google.com/open?id=1X3rcS5lBx8-TYWLlrWRVVz03oEF0XUR9
Please download from the above link. After unzip, please replace the zImage and rk3288-miniarm.dtb which are located at your micro sdcard.

plchampigny, 2018-09-20:
@jamess-huang Thank you! Can this be done while the device is running by changing those files on the /boot partition or this will not work?

jamess-huang, 2018-09-26:
Sorry for the late reply. It can put those files into /boot to replace the original ones.

plchampigny, 2018-09-27:
@jamess-huang Thank you, I did this and I think it worked, the os is booting. 

I got another issue right now when copying my image on another sd card and trying to boot it on a tinkerboard s (my initial image was on 2.0.4, made on tinkerboard (not s)). The os wont boot or show any image. I dont know how to debug this.

jamess-huang, 2018-09-28:
oh oops, it's my mistake, update kernel might cause the wifi do not work, because the kernel can't load the correct path of the wifi module. Please use the below deb file to install.
https://drive.google.com/open?id=1GX_MHCsks0Duh5GMViBLhyJkVUU-srI7

$ sudo dpkg -i kernel+modules+wlan_2.0.6_all.deb

closed 2018-11-21 by jamess-huang