TinkerBoard/debian_kernel #22: Chromium version 69 crashing on open.

alpagot, 2018-09-17:
```
chromium -version
Chromium 69.0.3497.81 built on Debian 9.5, running on Debian 9.5

```

Removing the version and installing a version such as `chromium_67.0.3396.87-1~deb9u1_armhf.deb` seems to fix the issue. Please investigate or advise what further info is needed. 

This is using `tinker-board-linaro-stretch-alip-v2.0.7.img` with apt-get update & upgrade run after a clean re-image of the SD card.

TinkerTeam, 2018-09-25:
Hi, 
Somehow the Debian's chromium package was broken.
Since some of user also have feedback, the Ubuntu's chromium package can successful ruining (tested with 68). So this problem believed is depend on Debina's pre-build issue.

And currently we have no good idea to let Debian watch this case.

Thanks.

ShinyMark, 2019-01-04:
Still happening as of January 2019. Flash the latest ROM, boot, apt update && upgrade, Chromium does not work. Installing firefox-esr after this also crashes instantly on boot.

TinkerTeam, 2019-01-07:
Debian still hasn't fixed them...
bug id: 906548, 906588, 909635, 913010, 914313
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=906548

dreamcarrior, 2019-07-17:
As of July 2019, I still have issue not able to use latest Chromium in TinkerOS 2.0.8