TinkerBoard/debian_kernel #24: Touch support is not working on RPI Display

sobrinho, 2018-09-24:
Hi there!

I can see and use a mouse/keyboard but no touch support.

Any to fix this?

sobrinho, 2018-09-24:
Apparently the version available on BR page is outdated.

https://www.asus.com/br/Single-Board-Computer/Tinker-Board/ vs https://www.asus.com/Single-Board-Computer/Tinker-Board/

Closing, it is probably that, will reopen if the issue still remains.

sobrinho, 2018-09-24:
Nope, same issue.

jamess-huang, 2018-09-25:
Hi, the RPI display you mean is the below one ?
https://www.raspberrypi.org/products/raspberry-pi-touch-display/

If so, the touch function should be work well. May I know your image version ? please use the below cmd to check. $ cat /etc/version

sobrinho, 2018-09-25:
Yes, this one.

Using the latest image [here](https://www.asus.com/Single-Board-Computer/Tinker-Board/) (14.1.2.50) it doesn't work, using the previous one (13.14.0.95) it does work.

But even with 13.14.0.95, if the app is shown in portrait mode, the touch is still on landscape mode (practically unusable in this case as I have to guess where to press).

I can make a video if you want to see both behaviors :)

jamess-huang, 2018-09-26:
Thanks for your information. We will check it.

jamess-huang, 2018-09-26:
Hi, Sobrinho
    Since this is an Android 7.1.2 issue, can you please close this one and open a new one under the Android 7.1.2_kernel git, thanks.
    https://github.com/TinkerBoard/android-7.1.2_kernel

jamess-huang, 2018-09-26:
Hi, Sobrinho
    We just checked again, we can't duplicate the issue. 
    Did you modify any android settings ? and did you ever try to re-flash 13.14.0.95 image and test again ?

sobrinho, 2018-09-26:
Sure, I tried that many times.

All applications are presenting this weird behavior when they are in portrait mode.

The tinker board is obviously in landscape but some applications like Waze and Spotify forces you to use the portrait mode to log in and that's when the image turns around (as expected) but the touch position is like in the wrong way.

I will make a video of that and post here, just a sec.

Btw, I haven't modified any configuration, just flashed the img from the downloads page directly into the Tinker Board S.

I doubt that but it may related to the S instead of the regular version of Tinker Board?

sobrinho, 2018-09-26:
You can see in this video that it is working as expected when in landscape mode but when I open the Google Calendar and it forces the OS to be in portrait mode, it keeps the touch working in landscape.

You can see that if I "guess" what the position would be in landscape mode, it works, but you can imagine how is that hard when you have to press a specific button in the screen.

This is happening to all applications that forces to be in portrait modes like Google Calendar, Spotify and Waze.

It is not the end of the world since those screens are more common during the login or a help but it is difficult to operate the screen like that.

https://youtu.be/oBKFvZK7v0Y

TinkerTeam, 2018-09-27:
Hi @sobrinho,

We will check it ASAP. 
But it looks like is the TinkerOS_Android_v14.1.2.50's problem.
Could you please kind to re-create a new issue at right place, [TinkerBoard/android-7.1.2_kernel](https://github.com/TinkerBoard/android-7.1.2_kernel)

Thanks.

sobrinho, 2018-09-27:
Hi there @TinkerTeam!

The problem is when I'm running the v13.14.0.95, I found that v14 is not working the touchscreen at all.

I can open a separated issue for that but I'm hoping that we can fix that problem of rotating the touchscreen here (the issue in the video).

[image]

TinkerTeam, 2018-09-27:
Hi @sobrinho, 

Okay, so also have the issue on Android 6. (after rotating)
Could you help to open anther one at [TinkerBoard/android-6.0.1_kernel](https://github.com/TinkerBoard/android-6.0.1_kernel) too?

sobrinho, 2018-09-27:
Ops, sorry about that!

Sure, thanks!

closed 2018-09-27 by sobrinho