TinkerBoard/debian_kernel #26: fatal error: openssl/bio.h: No such file or directory

sellep, 2018-10-21:
After checking branch linux4.4-rk3288 and creating the architectures default config, the compiling results in the following error:
make ARCH=arm CROSS_COMPILE=/usr/bin/arm-linux-gnueabi- -j5 zImage
  CHK     include/config/kernel.release
  CHK     include/generated/uapi/linux/version.h
  CHK     include/generated/utsrelease.h
  HOSTCC  scripts/extract-cert
  CC      scripts/mod/empty.o
  CC      scripts/mod/devicetable-offsets.s
scripts/extract-cert.c:21:10: fatal error: openssl/bio.h: No such file or directory
 #include <openssl/bio.h>

sellep, 2018-10-21:
Installing libssl-dev (WSL, Ubuntu) fixed it for me ...

closed 2018-10-21 by sellep

starjasmine, 2019-12-05:
@sellep It works for me as well. Thanks :+1:

helviorhild, 2020-03-02:
On build root you must enable CA_CERTIFICATES, for include ssl libraries. .

Licaifei, 2020-05-07:
It works,Thanks

tamim8, 2020-07-06:
It works,Thanks

tamim8, 2020-07-06:
plz help me....

 crypto/ghash-generic.o
  CC      crypto/asymmetric_keys/asymmetric_type.o
/home/ubuntu/Desktop/kernel/redmi7/kernel/time/hrtimer.c: In function '__remove_hrtimer':
/home/ubuntu/Desktop/kernel/redmi7/kernel/time/hrtimer.c:891:1: warning: label 'out' defined but not used [-Wunused-label]
error, forbidden warning: hrtimer.c:891
make[3]: *** [/home/ubuntu/Desktop/kernel/redmi7/scripts/Makefile.build:339: kernel/time/hrtimer.o] Error 1
make[2]: *** [/home/ubuntu/Desktop/kernel/redmi7/scripts/Makefile.build:653: kernel/time] Error 2
make[1]: *** [/home/ubuntu/Desktop/kernel/redmi7/Makefile:1141: kernel] Error 2

peterwoo339, 2022-10-13:
> libssl-dev

BTW, on centos it would be openssl-devel