TinkerBoard/debian_kernel #27: 4.4.103+ headers are wrong

Ransetsu, 2018-11-08:
So I installed Tinkeros_debian v2.0.7 and installed the headers for it by doing `dpkg -i <headerfile.deb>` on the deb file you guys provide, but all the executables in /usr/src/linux-headers-4.4.103+/scripts are 64-bit executables when the OS is 32-bit. Both the OS and the headers are fresh installs. As a result I can't run "make" to compile anything. "make" results in a "gcc: error: unrecognized command line option '-mgeneral-regs-only' and "make ARCH=arm" errors out with "/bin/sh: 1: ./scripts/recordmcount: Exec format error".

I was able to fix this by going to "/usr/src/linux-headers-4.4.103+" and running `make ARCH=arm scrips` which threw an error, but seems to have recompiled all the executables in scripts. Then I was finally able to compile by running `make ARCH=arm`. Just running `make` still does not work. Anyway, I'm not exactly sure what is wrong here, but I don't think this is working as intended.

jamess-huang, 2018-11-09:
Hi, 
   As I know, there should be little executable files in the kernel header, all of them should be shell script, perl, python script and so many header files. May I know why you need to run the executable files in the kernel headers? It should be no necessary to run/recompile the kernel header files if you just want to build your own kernel modules.

Ransetsu, 2018-11-09:
I'm not explicitly trying to run them. when i try and run `make ARCH=arm` to build the pf_ring kernel module it tries to run recordmcount (which is in /usr/src/linux-headers-4.4.103+/scripts) and fails because it is not a 32-bit executable.
 If you would like the try and reproduce what I did:
fresh v2.0.7 os install and run:
`apt update && apt upgrade -y && apt full-upgrade`
`wget https://github.com/TinkerBoard/debian_kernel/releases/download/2.0.7/linux-headers-4.4.103+_4.4.103+-1_armhf.deb`

`dpkg -i linux-headers-4.4.103+_4.4.103+-1_armhf.deb`
`apt install build-essential bison flex libnuma-dev`
`git clone https://github.com/ntop/PF_RING.git`
`cd PF_RING/kernel`
`make` *which errored out with "gcc: error: unrecognized command line option '-mgeneral-regs-only"
so i tried: 
`make ARCH=arm` *which erroed out with "/bin/sh: 1: ./scripts/recordmcount: Exec format error" because it was a 64-bit executable.
so then i recompiled everything in /usr/svr/linux-headers-4.4.103+/scripts" to 32-bit with:
`cd /usr/src/linux-headers-4.4.103+ && make ARHC=arm scripts`
and after that i was able to build the pf_ring kernel module with:
`make ARCH=arm` 
`make` on it's own is not compiling things correctly for some reason.

jamess-huang, 2018-11-09:
Thanks. We will update a new header files for 32 bit.

jamess-huang, 2018-11-21:
The kernel header has been updated.
https://github.com/TinkerBoard/debian_kernel/releases/download/2.0.7/linux-headers-4.4.103+_4.4.103+-3_armhf.deb

closed 2018-11-21 by jamess-huang