TinkerBoard/debian_kernel #34: Flash tinkerboard image on tinkerboard s

plchampigny, 2019-01-09:
I've tried to flash a tinkerboard image (2.0.7) on a tinkerboard s. The image has been read using win32diskimager from a modified version of the tinkerboard (only apps have been installed). If I flash this image on the eMMC or on a sd card, the tinkerboard s does not boot (or at least doesnt send anything to the hdmi). If I flash that image on an sd card and put it in a 1st gen tinkerboard, it boots up properly.

Is there something I have to do to fix my image for tinkerboard s?

plchampigny, 2019-01-10:
I do believe that my AC adapter is a 2.5A/5V. I'll check tomorrow at work. Meanwhile I can tell you I plugged the same sd card that is working on a tinkerboard on a tinkerboard s and I used the same ac, hdmi, etc and its not working.

jamess-huang, 2019-01-10:
When the tinker S is un-boot, may I know the 3 LED status ? Is the red LED on and is the yellow LED blinking ?

krystofurr, 2019-01-24:
I purchased a 1st gen Tinkerboard recently and tried to use the v2.07 image.  Got a similiar result.  No boot.  Black screen using HDMI connection.  I only have a red LED on myself.

jamess-huang, 2019-01-25:
Hi, krystofurr
    Can you see the yellow LED blink sometimes ?

krystofurr, 2019-01-25:
Yes, occassionally they do.  I tried v1.8 and it seems to work fine.  Funny that the main website doesn't have the latest version for download.  I downloaded the image from here. https://www.asus.com/ca-en/Single-Board-Computer/Tinker-Board/

krystofurr, 2019-01-25:
I'll try to find the exact models.  The first was a 32 inch Sony bravia lcd tv ( works fine with an rpi 3 and v1. 8 of the os).  The second was a 50 inch insignia led tv (Also works with rpi 3 and v1. 8 of the os). When it did come up with 1.8 though the resolution is very off and if u disconnect / reconnect the HDMI it fixes a little.  Also there is a high static noise that is constant. This only happens with the sony bravia

krystofurr, 2019-01-25:
Got them.

Sony Bravia KDL-32ML130
Insignia NS-50L260A13