TinkerBoard/debian_kernel #35: Camera IMX219 not working with OPENCV

Yuehching, 2019-03-04:
Im using camera IMX219 on thinkerOS 2.0.8. I followed instructions on https://tinkerboarding.co.uk/wiki/index.php/CSI-camera-2.0.8 and it shows 
```
entity 7: imx219 2-0010 (1 pad, 1 link)
            type V4L2 subdev subtype Sensor flags 0
            device node name /dev/v4l-subdev2
	pad0: Source
		[fmt:SRGGB10_1X10/1920x1080 field:none]
		-> "rockchip-sy-mipi-dphy":0 [ENABLED]
```

after command
`media-ctl -p`
things just appear normal
but when I test the camera with the following code, things go wrong. 

```
#include <opencv2/highgui.hpp>
#include <iostream>
#include <opencv2/opencv.hpp>
using namespace cv;
using namespace std;

int main(int argc, char** argv)
{

    VideoCapture cap;

    cap = VideoCapture(0);
    if(!cap.isOpened())
    {
        cout << "Couldn't find camera: " << endl;
        return -1;
    }

    for(;;)
    {
        Mat frame;
        cap >> frame; // get a new frame from camera/video or read image

        if (frame.empty())
        {
            waitKey();
            break;
        }

        imshow("image", frame);
        if (waitKey(1) >= 0) break;
    }

    return 0;
}
```

the result shows:
```
VIDEOIO ERROR: V4L2: Unable to capture video memory.VIDEOIO ERROR: V4L: can't open camera by index 0
/dev/video0 does not support memory mapping
```

I tried many ways but cannot find a solution...before this, my raspberry pi 3B+ works well with IMX219 and opencv.

topdjgod, 2019-03-04:
You need to compile opencv to enable v4l support.
Please see below link:
https://tinkerboarding.co.uk/wiki/index.php/CSI-camera-2.0.8

Yuehching, 2019-03-05:
> rt.
> Please see bel

I tried the c++ sample code in the link you give. and it always spit "Couldn't find camera". It's widely known that raspberry pi  CSI camera cannot be directly used by VideoCapture() in opencv unless executing `modprobe bcm2835-v4l2` ahead. So I think tiner board is probably the same. I want to know if there is a similar way to make CSI camera accessible by opencv...

topdjgod, 2019-03-27:
@Yuehching 
First, make sure your camera is good. Please use test-camera.sh in TinkerOS v2.0.8 to test your camera. Second, install opencv. Please see "OpenCV in python (Face Detection)" to install opencv.
Then you can try c++ sample code.