TinkerBoard/debian_kernel #40: Audio frames are dropped during on-ongoing record session "retire_capture_urb"

shaycohen, 2019-03-25:
When recording an on-going Audio session, using on-board audio device, kernel log shows: 
"retire_capture_urb: NN callbacks suppressed"
During this time-frame, the produced audio file is missing frames. 

Reproduced on 
TinkerOS versions 2.0.7 and 2.0.8
Kernel versions: 4.4.103+ and 4.4.132+

jamess-huang, 2019-03-25:
Can you give us a detail configuration and the duplicate step (including the recording command/APP)

shaycohen, 2019-03-25:
Hi @jamess-huang , Thanks for the prompt reply. 
I've attached alsa.conf (as alsa.conf.log in order to comply to GitHub's allowed extensions policy)
[alsa.conf.log](https://github.com/TinkerBoard/debian_kernel/files/3002271/alsa.conf.log)

The recording code snippet is the following (Python3 code): 

import time, redis 
class Record():
    def __init__(self):
        self.r = redis.StrictRedis(unix_socket_path="/var/run/redis/redis.sock")

    def record_callback(self, in_data, frame_count, time_info, status):
        self.r.xadd("DATA", { b'timestamp': time_info['input_buffer_adc_time'], 'pcm': in_data }, maxlen=2000)
        return in_data, self.paContinue

    def record(self):
        import pyaudio
        self.paContinue = pyaudio.paContinue
        audio = pyaudio.PyAudio()
        audioFormat = 8
        channels = 1
        rate = 44100
        index = 1
        chunk = 4096
        self.r.set('START_TS', int(time.time()))
        device=audio.get_device_info_by_host_api_device_index(0,index)
        self.r.set('RECORDER_NAME',device.get('name'))
        stream = audio.open( format=audioFormat, channels=channels, rate=rate, input=1,
        input_device_index=index, frames_per_buffer=chunk, stream_callback=self.record_callback)
        self.r.set("AUDIO_LATENCY", stream.get_input_latency())
        stream.start_stream()
        while True:
            time.sleep(10)
        stream.stop_stream()
        stream.close()
        audio.terminate()
        return
rec = Record()
rec.record()

jamess-huang, 2019-03-25:
Sorry to ask this: Did you try to use the same python code on Raspberry Pi or another SBC board ?
Just make sure that there is no issue in this python code.

shaycohen, 2019-04-02:
I've setup a test environment on an RPi 3B+, using an external / USB audio device (as the RPi onboard does not have a capture interface). 
Kernel version: 4.19.25-v7+
This device does not seem to have any callback drops nor missing audio chunks on the recorded data.

rotemtim, 2019-04-22:
Hi
I've just encountered the same problem with tinkerOS 2.0.8, there are a lot of audio chunks drops.
@james-huang did you find any resolution to this problem?

jamess-huang, 2019-04-23:
Hi, 
    We just got one usb ODD. We will check this issue and keep update.

shaycohen, 2019-04-28:
Hi @jamess-huang 
Will you please post a short update / ETA about this ? 
Thank you very much for your support, 
Shay

topdjgod, 2019-05-02:
@shaycohen
Could you help to try below sample code to check if it can reproduce?
We use below sample code can't reproduce your issue.

import pyaudio
import wave

CHUNK = 4096
FORMAT = 8
CHANNELS = 1
RATE = 44100
RECORD_SECONDS = 60
WAVE_OUTPUT_FILENAME = "output.wav"
p = pyaudio.PyAudio()
stream = p.open(format=FORMAT,channels=CHANNELS,rate=RATE,input=True,frames_per_buffer=CHUNK)
print("*recording")
frames = []
for i in range(0,int(RATE/CHUNK*RECORD_SECONDS)):
    data = stream.read(CHUNK)
    frames.append(data)
print("*done recording")
stream.stop_stream()
stream.close()
p.terminate()
wf = wave.open(WAVE_OUTPUT_FILENAME, 'wb')
wf.setnchannels(CHANNELS)
wf.setsampwidth(p.get_sample_size(FORMAT))
wf.setframerate(RATE)
wf.writeframes(b"".join(frames))
wf.close()

@anyone
Do you have a simple way to reproduce the problem?

shaycohen, 2019-05-02:
@topdjgod 
Thank you for your support. 
The proposed code does not use a callback method for capturing audio, thus no callbacks would ever be missed. 
This method of reading (sequential stream.read) is not good enough when you need to know the exact time stamp of the captured data (normally provided by the audio driver to the callback function as an argument). 

Will you please try to reproduce with the code sample I have posted above ?

topdjgod, 2019-05-03:
@shaycohen
[recording_redis.zip](https://github.com/TinkerBoard/debian_kernel/files/3140507/recording_redis.zip)
Use your sample code as attached file with redis 5.0.4 to test, but still not reproduce it.
Could you help to try the attached sample code again?

ps. redis 5.0.4 default the position of redis.sock is in /tmp