TinkerBoard/debian_kernel #42: RTL8812au wifi-direct not advertising

visbs, 2019-04-15:
Hi,

I am using RTL8812AU based chipset USB Wi-Fi Adapter (Like ASUS AC56 , Linksys WUSB6300) in ubuntu (tried in different OS and Kernel version).
And I am trying to make use of WI-Fi Direct using wpa_supplicant (P2P Enabled) with user app wpa_cli.
In case of Intel Wireless Driver, whenever I do P2P_FIND or any scan command, from wpa_cli, my device is visible in any Wi-Fi Direct supported devices like Android, Windows etc.
But in case of Realtek driver (rtl8812au), my device is not visible or listed in Wi-Fi Direct device list, whereas I can get the nearby peers available in my device. As well I am able to connect to any SSID available.
So, am I missing something additional setup required for driver to make my device visible to other peer?
Any help would be very helpful?
PS: I did go through rtl8723 issue but the changes there didn't help.
Thank you,

Sam2Hsu, 2019-04-18:
Hi visbs,

I am using D-LINK DWA-182 on TinkerOS 2.0.8(Debian 9). It can find p2p devices by wpa_cli P2P_FIND.

First, let network manager unmanaged your USB Wi-Fi dongle. Add **unmanaged-devices** to your /etc/NetworkManager/NetworkManager.conf
```
[main]
plugins=ifupdown,keyfile

[ifupdown]
managed=false

[device]
wifi.scan-rand-mac-address=no

[keyfile]
unmanaged-devices=mac:f8:e9:03:da:9c:b1
```

Then, create **/etc/wpa_supplicant/p2p_supplicant.conf** with following content.
```
ctrl_interface=/var/run/wpa_supplicant
update_config=1
device_name=tinkerboard
device_type=1-0050F204-1
p2p_go_intent=15
p2p_go_ht40=1
```
I assume new interface name is **wlan1**.
You can find P2P devices with below commands.
```
iwpriv wlan1 p2p_set enable=3
wpa_supplicant -c /etc/wpa_supplicant/p2p_supplicant.conf -i wlan1 &
wpa_cli -i wlan1 p2p_find
wpa_cli -i wlan1 p2p_peers
```

Thanks,
Sam

visbs, 2019-04-19:
Sorry for the delay andThanks for the update will  try it.

visbs, 2019-04-22:
@Sam2Hsu  even using my USB device i can find the peers, but in android devices am not able to see(i.e the driver is not advertising).

Sam2Hsu, 2019-04-23:
Hi Visbs,

My dongle can be searched by android devices after become group owner.
I am using TinkerOS_2.0.8. and the same steps with NetworkManager and p2p_supplicant.conf.

Here's the command to be group owner.
```
iwpriv wlan1 p2p_set enable=3
wpa_supplicant -c /etc/wpa_supplicant/p2p_supplicant.conf -i wlan1 &
wpa_cli -i wlan1 p2p_group_add persistent
wpa_cli -i wlan1 wps_pbc
```
Dongle can be searched after p2p_group_add, and can be connected after enter wps push button.
BTW, p2p device name will be your dongle mac address.
[Android Wi-Fi Direct page screenshot](https://i.imgur.com/EZ4Z7Xf.jpg)

Thanks,
Sam

closed 2020-01-03 by visbs