TinkerBoard/debian_kernel #48: Audio Injector Octo S.O.S.

fiscalfreddy, 2019-08-07:
Hello fellow kernel harvesters,

I've just compiled the Debian kernel 2.0.10 after downloading the .zip file from the release branch of the repo.  I have tried compiling several times.  I'm using an Ubunto 64bit (VirtualBox) computer and following these build instructions  

https://gist.github.com/TinkerTeam/6286550ce70d34f6b3d483cd803da786

However, after I mount my sd card and boot I'm not seeing the audioinjector-addons.dtbo file show up in the /boot/overlays directory, and the card doesn't seem to work.

Any suggestions? Perhaps I should try a different .config file?

Thanks!

--ff

jamess-huang, 2019-08-07:
Can you help to try the below instructions ?
https://tinkerboarding.co.uk/wiki/index.php/Software#Kernel

fiscalfreddy, 2019-08-07:
Thanks for your help! After that compile, the tinkerboard is now recognizing the device.  I haven't gotten playback to work yet but this is still promising.

I did have to manually update the boot partition of the sd card to include the audioinjector's .dtbo file in addition to those instructions.

jamess-huang, 2019-08-08:
Yes. We just add the overlay in the overlays folder. User needs to include the overlay by adding "intf:dtoverlay=xxx" into the /boot/hw_intf.conf file.

closed 2019-08-08 by jamess-huang