TinkerBoard/debian_kernel #50: dist-upgrade

p1r473, 2020-03-12:
Is there any way to upgrade from 2.0.8 to 2.0.11 without having to do a fresh install?????
dist-upgrade doesnt do anything...

igorpecovnik, 2020-03-19:
Manual only. Armbian supports upgrading since the day zero https://www.armbian.com/tinkerboard/

dvergeylen, 2020-05-05:
For the ones like me wondering where to download the latest image (Release page here is stuck at 2.0.8) here is the link:
https://www.asus.com/uk/Single-Board-Computer/Tinker-Board/HelpDesk_Download/

But no clue where the kernel configuration source code lies... :confused:

igorpecovnik, 2020-05-05:
No, kernel configuration is not there. That is Armbian's config which has nothing to do with TinkerOS.

dvergeylen, 2020-05-05:
@igorpecovnik, thank you for the clarification 🙂

TinkerTeam, 2020-05-05:
here is the kernel config: https://github.com/TinkerBoard/debian_kernel/blob/develop/arch/arm/configs/miniarm-rk3288_defconfig

but 2.0.11 version not release on Github yet, we would upload it asap.

p1r473, 2020-05-05:
@TinkerTeam Please make dist-upgrade (or similar) work.... I hate reflashing after every single new version comes out :(
Would be nice to auto upgrade with one command...
I reflash from 2.0.7 to 2.0.8
I reflash 2.0.8 to 2.0.11
I reflash 2.1.11
Takes so much time!

igorpecovnik, 2020-05-05:
> Please make dist-upgrade

That is a very expensive feature which is unlikely you will ever get. Check Armbian. It has it all.

> I had trouble getting my TFT screen to work with Armbian so I am back on Tinkerboard

It is always a trade. In case you want to get your screen working, join here https://github.com/armbian/build/pull/1887

p1r473, 2020-05-06:
They had said a long time ago they had it working, or would have it working soon. Hopefully, they will!

igorpecovnik, 2020-05-06:
> They had said a long time ago they had it working, or would have it working soon. Hopefully, they will!

Armbian with legacy kernel has both - the same level of hardware support and normal updating procedure. But I still recommend you to switch to the modern kernel for at least security reasons. We are fixing this feature now and we will not go back when you wake up from the dream.

dvergeylen, 2020-05-19:
> but 2.0.11 version not release on Github yet, we would upload it asap.

@TinkerTeam, do you plan to release it soon? 😇

dvergeylen, 2020-09-09:
Hi @TinkerTeam,

>but 2.0.11 version not release on Github yet, we would upload it asap.

Any updates on this?