TinkerBoard/debian_kernel #51: SPI transfer size >32 bytes corrupts memory and will likely hang the system

dhendrix, 2020-06-04:
When transferring >32 bytes on SPI0 or SPI2, it seems to clobber memory and will likely crash the system. I noticed this when trying to use the SPI buses to read a flash chip using transfer size == 4096 bytes as per /sys/module/spidev/parameters/bufsiz. Using 32 bytes instead seems to work fine.

Using Linux's spidev_test tool (under linux/tools/spi), we can easily reproduce this behavior. Examples:
1. Transfer 32 bytes - We see the correct bytes on TX (16 a's followed by 16 b's in this example), and all 0xff's on RX as expected
linaro@tb1:~/src/linux/tools/spi$ sudo ./spidev_test -v -D /dev/spidev0.0 -s 1000000 -i /tmp/in_32b.bin -o /tmp/out_32b.bin
spi mode: 0x0
bits per word: 8
max speed: 1000000 Hz (1000 KHz)
TX | 61 61 61 61 61 61 61 61 61 61 61 61 61 61 61 61 62 62 62 62 62 62 62 62 62 62 62 62 62 62 62 62  | aaaaaaaaaaaaaaaabbbbbbbbbbbbbbbb
RX | FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  | ................................

2. Transfer 64 bytes - We see the correct bytes on TX (16 a's followed by 16 b's, 16 c's, and 16 d's). However, RX shows corruption immediately.
linaro@tb1:~/src/linux/tools/spi$ sudo ./spidev_test -v -D /dev/spidev0.0 -s 1000000 -i /tmp/in_64b.bin -o /tmp/out_64b.bin
spi mode: 0x0
bits per word: 8
max speed: 1000000 Hz (1000 KHz)
TX | 61 61 61 61 61 61 61 61 61 61 61 61 61 61 61 61 62 62 62 62 62 62 62 62 62 62 62 62 62 62 62 62  | aaaaaaaaaaaaaaaabbbbbbbbbbbbbbbb
TX | 63 63 63 63 63 63 63 63 63 63 63 63 63 63 63 63 64 64 64 64 64 64 64 64 64 64 64 64 64 64 64 64  | ccccccccccccccccdddddddddddddddd
RX | FF 00 00 00 C6 88 DE BE 00 00 00 00 00 00 00 00 2F 74 6D 70 2F 72 61 6E 64 6F 6D 5F 36 34 62 5F  | ....ƈ޾......../tmp/random_64b_
RX | 6F 75 74 2E 62 69 6E 00 2F 6C 69 62 63 2E 73 6F 2E 36 00 00 2E 62 69 6E 00 6E 69 74 6F 72 65 2D  | out.bin./libc.so.6...bin.nitore-

ivbsd1, 2020-07-12:
Hi,
I have the same problem with Tinker Board running Debian:
Linux tinkerboard 4.4.132+ #1 SMP Wed Aug 21 19:15:55 CST 2019 armv7l GNU/Linux

Are there some updates about this SPI problem ? When fix expected ?
Maybe someone can suggest workaround ?
Thanks

dcarlson27, 2022-04-22:
I just ran in to this.   Can anyone share possible fix, or when spidev was updated to address this?

yishinhung, 2022-04-28:
Hi @dcarlson27 :
     Which image version do you use? I use the following image version and verify normally.
* [Tinker Board S Debian 10 V3.0.11](https://tinker-board.asus.com/download-list.html?product=tinker-board-s#download20211110):

- Test by spidev_test tools
root@linaro-alip:/home/linaro/Desktop/spidev-test# ./spidev_test -v -D /dev/spidev0.0 -s 1000000 -i ../in_64.bin -o out_64.bin
spi mode: 0x0
bits per word: 8
max speed: 1000000 Hz (1000 KHz)
TX | 61 61 61 61 61 61 61 61 61 61 61 61 61 61 61 61 62 62 62 62 62 62 62 62 62 62 62 62 62 62 62 62  | aaaaaaaaaaaaaaaabbbbbbbbbbbbbbbb
TX | 63 63 63 63 63 63 63 63 63 63 63 63 63 63 63 63 64 64 64 64 64 64 64 64 64 64 64 64 64 64 64 64  | ccccccccccccccccdddddddddddddddd
RX | FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  | ................................
RX | FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  | ................................
root@linaro-alip:/home/linaro/Desktop/spidev-test#
root@linaro-alip:/home/linaro/Desktop/spidev-test# ./spidev_test -v -D /dev/spidev0.0 -s 1000000 -i ../in_32.bin -o out_32.bin
spi mode: 0x0
bits per word: 8
max speed: 1000000 Hz (1000 KHz)
TX | 61 61 61 61 61 61 61 61 61 61 61 61 61 61 61 61 62 62 62 62 62 62 62 62 62 62 62 62 62 62 62 62  | aaaaaaaaaaaaaaaabbbbbbbbbbbbbbbb
RX | FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  | ................................