TinkerBoard/debian_kernel #56: Build debian_kernel fails

sv1eia, 2021-03-11:
Hi,
Using Ubuntu Focal virtual machine for all Linux compiles and trying to build the Tinkerboard debian_kernel
following these
https://tinker-board.asus.com/doc_tb.html#os/
and
https://tinkerboarding.co.uk/wiki/index....ource_code

but it fails upon execution of
make zImage ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf-

  CC      arch/arm/kernel/perf_callchain.o
  CC      arch/arm/kernel/perf_event_xscale.o
  CC      arch/arm/kernel/perf_event_v6.o
  CC      arch/arm/kernel/perf_event_v7.o
  CC      arch/arm/kernel/topology.o
  CC      arch/arm/kernel/vdso.o
arch/arm/kernel/vdso.c: In function ‘vdso_init’:
arch/arm/kernel/vdso.c:182:6: warning: ‘memcmp’ reading 4 bytes from a region of size 1 [-Wstringop-overflow=]
error, forbidden warning: vdso.c:182
make[1]: *** [scripts/Makefile.build:278: arch/arm/kernel/vdso.o] Error 1
make: *** [Makefile:1028: arch/arm/kernel] Error 2

christos@ubuntu:~/tinkerboard/debian_u-boot$ uname -a
Linux ubuntu 5.8.0-44-generic #50~20.04.1-Ubuntu SMP Wed Feb 10 21:07:30 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

Are there any specific instructions on what prerequisites are needed, on what platform and how to build the debian_kernel ?

asus-leslieyu, 2021-03-24:
There are issues to build with Ubuntu Focal. Please use the following Docker environment to build with Ubuntu Xenial. Thanks a lot!

https://github.com/asus-iot/asus-docker-env

sv1eia, 2021-03-31:
Installed docker in Focal and followed the instructions in https://github.com/asus-iot/asus-docker-env

When I tried a new compile inside the docker container got the same error

  CC      arch/arm/kernel/perf_event_xscale.o
  CC      arch/arm/kernel/perf_event_v6.o
  CC      arch/arm/kernel/perf_event_v7.o
  CC      arch/arm/kernel/topology.o
  CC      arch/arm/kernel/vdso.o
arch/arm/kernel/vdso.c: In function 'vdso_init':
arch/arm/kernel/vdso.c:182:6: warning: 'memcmp' reading 4 bytes from a region of size 1 [-Wstringop-overflow=]
error, forbidden warning: vdso.c:182
make[1]: *** [scripts/Makefile.build:278: arch/arm/kernel/vdso.o] Error 1
make: *** [Makefile:1028: arch/arm/kernel] Error 2
christos@c36f5cb3ff41:/source/debian_kernel$ 

Is there any step by step guide on how to use the aforementioned docker environment in order to build the tinkerboard uboot and kernel?

asus-leslieyu, 2021-03-31:
Hi,

Did you use the following command to change the Dockfile to the one from xenial? The default one is from focal. Please change it to from xenial for Tinker Board (S). Thanks a lot!

asus_docker_env_set_dockerfile ./xenial/Dockerfile

sv1eia, 2021-04-01:
As I mentioned already, the build system is Ubuntu Focal, so according to https://github.com/asus-iot/asus-docker-env I do not need to switch to Xenial

christos@ubuntu:~/tinkerboard/asus-docker-env$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.2 LTS
Release:	20.04
Codename:	focal
christos@ubuntu:~/tinkerboard/asus-docker-env$ 

Do you want me to regardless try to switch the docker to xenial with 
asus_docker_env_set_dockerfile ./xenial/Dockerfile

??

sv1eia, 2021-04-01:
Ok, switched the docker container to xenial and now kernel compiles just fine.
Thanks for the help.

closed 2021-04-01 by sv1eia

asus-leslieyu, 2021-04-01:
Yes, regardless the host is focal or not, need to use Dockerfile based on xenial since there are issues related to tool chains from focal with the source code.