TinkerBoard/debian_kernel #57: About newer releases of kernel

Zaryob, 2021-03-19:
Is there any way to use kernel 5.x in tinkerboard 2?
Are you planning to move it newer releases of kernel 5.x

theos10, 2021-04-05:
Same question.
How tu upgrade kernel ?
Thank.

dvergeylen, 2021-04-05:
You have to configure and compile it yourself, see here: https://github.com/TinkerBoard/debian_kernel/issues/38#issuecomment-565722671

In addition to that you can use the Docker environment: https://github.com/TinkerBoard/debian_kernel/issues/56#issuecomment-805441822

Zaryob, 2021-04-05:
> You have to configure and compile it yourself, see here: [#38 (comment)](https://github.com/TinkerBoard/debian_kernel/issues/38#issuecomment-565722671)
> 
> In addition to that you can use the Docker environment: [#56 (comment)](https://github.com/TinkerBoard/debian_kernel/issues/56#issuecomment-805441822)

what about the hardware acceleration and gpu accelerated chromium. I tried to compile from scratch but not worked hw accel.

dvergeylen, 2021-06-03:
For crypto acceleration you need: `CONFIG_CRYPTO_DEV_ROCKCHIP` flag.

Zaryob, 2021-06-03:
> For crypto acceleration you need: `CONFIG_CRYPTO_DEV_ROCKCHIP` flag.

Thx

6a5h4, 2021-06-04:
Which driver needs to be activated in kernel config to support the eMMC HS200 on Tinkerboard S or 2S?

dvergeylen, 2021-06-04:
> Which driver needs to be activated in kernel config to support the eMMC HS200 on Tinkerboard S or 2S?

I guess it's `PHY_ROCKCHIP_EMMC`.

To find that kind of option, run `ARCH=arm make menuconfig` and use '/' to start search prompt. For instance, searching for 'emmc' gives:

```
Symbol: PHY_ROCKCHIP_EMMC [=n]
  │ Type  : tristate
  │ Prompt: Rockchip EMMC PHY Driver
  │   Location:
  │     -> Device Drivers
  │ (1)   -> PHY Subsystem
  │   Defined at drivers/phy/rockchip/Kconfig:12
  │   Depends on: ARCH_ROCKCHIP [=y] && OF [=y]
  │   Selects: GENERIC_PHY [=y]
  │
  │
  │ Symbol: PWRSEQ_EMMC [=y]
  │ Type  : tristate
  │ Prompt: HW reset support for eMMC
  │   Location: 
  │     -> Device Drivers
  │ (2)   -> MMC/SD/SDIO card support (MMC [=y])
  │   Defined at drivers/mmc/core/Kconfig:5
  │   Depends on: MMC [=y] && OF [=y]
```

`[=n]` means it's not enabled (in my `.config`). Find it at the specified location and enable it. You might have to make it a module, depending on custom tinker scripts.

closed 2022-09-19 by Zaryob