TinkerBoard/debian_kernel #61: OpenCL Performance issue help wanted

Saikatsaha1996, 2021-06-04:
Asus Tinker board
Os Tinker Debian 
rockchip 3288
Gpu Mali T 760
Gpu frequency max 600 mhz
But OpenCL GPU frequency showing only 5 mhz
How can I slove it ?
[image]

asusiot, 2021-09-08:
hello @Saikatsaha1996 
looks like it's a common issue in mali gpu: https://discuss.96boards.org/t/frequency-of-gpu/4946/4 

as someone mentioned in the thread, the gpu max freq is statically set to 5000; you can change the frequency in this line and see if it improves:
### #define GPU_FREQ_KHZ_MAX 5000

https://android.googlesource.com/kernel/hikey-linaro/+/android-hikey-linaro-4.14/drivers/gpu/arm_gpu/platform/hisilicon/mali_kbase_config_platform.h#26

Saikatsaha1996, 2021-09-08:
> hello @Saikatsaha1996
> looks like it's a common issue in mali gpu: https://discuss.96boards.org/t/frequency-of-gpu/4946/4
>
> as someone mentioned in the thread, the gpu max freq is statically set to [...]

Woww nice .. I will trying thank you...