TinkerBoard/debian_kernel #64: newly buyed tinker board S R2.0 not booting

sagar-k-ai, 2022-03-05:
Hi , I flashed  "Tinker_Board-Debian-Buster-v3.0.11-20211026" image in tinker board S R2.0 , after power on by 5v /3A supply ,board is not booting , after starting kernel many failed logs are coming. 
I tried different images from this link  https://github.com/TinkerBoard/debian_kernel/releases , but still its not working.

is it board issue or  image issue?

please check logs below.

Starting kernel ...

[    2.035349] vcc_flash: regulator get failed, ret=-517
[    2.040926] vcc18_dvp: regulator get failed, ret=-517
[    2.228712] rk-vcodec ff9a0000.vpu-service: could not find power_model node
[    2.238391] rk-vcodec ff9c0000.hevc-service: could not find power_model node
[    2.265843] tinker-mcu: send_cmds: send command failed, ret = -6, retry again!
[    2.273388] tinker-mcu: send_cmds: send command failed, ret = -6, retry again!
[    2.280861] tinker-mcu: send_cmds: send command failed, ret = -6, retry again!
[    2.288292] tinker-mcu: send_cmds: send command failed, ret = -6, retry again!
[    2.295760] tinker-mcu: send_cmds: send command failed, ret = -6, retry again!
[    2.303025] tinker-mcu: send_cmds: send command failed
[    2.308174] tinker-mcu: tinker_mcu_probe: init_cmd_check failed, -6
[    2.314927] tinker-mcu: send_cmds: send command failed, ret = -6, retry again!
[    2.322407] tinker-mcu: send_cmds: send command failed, ret = -6, retry again!
[    2.329840] tinker-mcu: send_cmds: send command failed, ret = -6, retry again!
[    2.337313] tinker-mcu: send_cmds: send command failed, ret = -6, retry again!
[    2.344782] tinker-mcu: send_cmds: send command failed, ret = -6, retry again!
[    2.352051] tinker-mcu: send_cmds: send command failed
[    2.357200] tinker-mcu: tinker_mcu_ili9881c_probe: init_cmd_check[-6] failed, 0
[    2.376569] rockchip-drm display-subsystem: failed to bind ff960000.dsi (ops 0xc0fa7244): -517
[    2.391432] mali ffa30000.gpu: Failed to get regulator
[    2.396592] mali ffa30000.gpu: Power control initialization failed
[    2.448219] rk_gmac-dwmac ff290000.ethernet: cannot get clock clk_mac_speed
[    2.461852] rk_gmac-dwmac ff290000.ethernet: rk_get_eth_addr: mac address: 04:42:1a:d3:e8:9f
[    3.490521] tinker-ft5406: tinker_ft5406_probe: wait connected timeout
[    3.617636] imx219 2-0010: Error -5 setting default controls
[    3.636885] rk_tsadcv2_temp_to_code: Invalid conversion table: code=4095, temperature=2147483647
[    3.665200] rockchip_cpuclk_pre_rate_change: Invalid rate : 500000000 for cpuclk
[    3.675903] cpu cpu0: failed to read out thermal zone (-22)
[    3.682539] cpu cpu0: failed to find power_model node
[    3.747373] rknandbase v1.2 2018-05-08
[    3.763579] rockchip-drm display-subsystem: failed to bind ff960000.dsi (ops 0xc0fa7244): -517
[    3.772833] mali ffa30000.gpu: Failed to get leakage
[    3.777841] mali ffa30000.gpu: Failed to get pvtm
[    3.784543] rockchip-pinctrl pinctrl: pin gpio4-30 already requested by wireless-wlan; cannot claim for ff0d0000.dwmmc
[    3.795259] rockchip-pinctrl pinctrl: pin-150 (ff0d0000.dwmmc) status -22
[    3.802061] rockchip-pinctrl pinctrl: could not request pin 150 (gpio4-30) from group sdio0-int  on device rockchip-pinctrl
[    3.813187] dwmmc_rockchip ff0d0000.dwmmc: Error applying setting, reverse things back
[    3.821163] devfreq ffa30000.gpu: Couldn't update frequency transition information.
[    3.854548] rockchip-drm display-subsystem: failed to bind ff960000.dsi (ops 0xc0fa7244): -517
[    3.864742] rockchip-drm display-subsystem: failed to bind ff960000.dsi (ops 0xc0fa7244): -517
[    3.874865] rockchip-drm display-subsystem: failed to bind ff960000.dsi (ops 0xc0fa7244): -517
[    3.885028] rockchip-drm display-subsystem: failed to bind ff960000.dsi (ops 0xc0fa7244): -517
[    3.895136] rockchip-drm display-subsystem: failed to bind ff960000.dsi (ops 0xc0fa7244): -517
[    3.905203] rockchip-drm display-subsystem: failed to bind ff960000.dsi (ops 0xc0fa7244): -517
[    3.915266] rockchip-drm display-subsystem: failed to bind ff960000.dsi (ops 0xc0fa7244): -517
[    3.925296] rockchip-drm display-subsystem: failed to bind ff960000.dsi (ops 0xc0fa7244): -517
[    3.935371] rockchip-drm display-subsystem: failed to bind ff960000.dsi (ops 0xc0fa7244): -517
[    3.945391] rockchip-drm display-subsystem: failed to bind ff960000.dsi (ops 0xc0fa7244): -517
[    3.955464]  rockchip-drm display-subsystem: failed to bind ff960000.dsi (ops 0xc0fa7244): -517
[    3.965532] rockchip-drm display-subsystem: failed to bind ff960000.dsi (ops 0xc0fa7244): -517
[    3.975582] rockchip-drm display-subsystem: failed to bind ff960000.dsi (ops 0xc0fa7244): -517
[    3.985636] rockchip-drm display-subsystem: failed to bind ff960000.dsi (ops 0xc0fa7244): -517
[    3.995681] rockchip-drm display-subsystem: failed to bind ff960000.dsi (ops 0xc0fa7244): -517
[    4.005703] rockchip-drm display-subsystem: failed to bind ff960000.dsi (ops 0xc0fa7244): -517
[    4.015750] rockchip-drm display-subsystem: failed to bind ff960000.dsi (ops 0xc0fa7244): -517
[    4.025817] rockchip-drm display-subsystem: failed to bind ff960000.dsi (ops 0xc0fa7244): -517
[    4.036118] i2c i2c-6: of_i2c: modalias failure on /hdmi@ff980000/ports
[    4.371700] get_framebuffer_by_node: failed to get logo,offset
[    4.377549] rockchip-drm display-subsystem: failed to show loader logo
[    4.871975] asoc-simple-card sound-simple-card: call_usermodehelper fail, ret=-2, status=0
[    4.880411] asoc-simple-card sound-simple-card: ASoC: DAPM unknown pin Headphones
[    4.892962] /testcase-data/phandle-tests/consumer-a: could not get #phandle-cells-missing for /testcase-data/phandle-tests/provider1
[    4.904879] /testcase-data/phandle-tests/consumer-a: could not get #phandle-cells-missing for /testcase-data/phandle-tests/provider1
[    4.916825] /testcase-data/phandle-tests/consumer-a: could not find phandle
[    4.923819] /testcase-data/phandle-tests/consumer-a: could not find phandle
[    4.930792] /testcase-data/phandle-tests/consumer-a: arguments longer than property
[    4.938449] /testcase-data/phandle-tests/consumer-a: arguments longer than property
[    4.955748] rkisp1: update sensor failed
[    5.031973] asoc-simple-card sound-simple-card: call_usermodehelper fail, ret=-2, status=0
[    5.040427] asoc-simple-card sound-simple-card: ASoC: DAPM unknown pin Headphones
[    5.716371] systemd[1]: Failed to find module 'autofs4'
[    5.733811] cgroup: cgroup2: unknown option "nsdelegate"

GillesC, 2022-04-15:
One advice, move away from Tinker OS. Mine wasn't working either. So I switched to Armbian: https://www.armbian.com/tinkerboard/. Which does work.

TinkerTeam, 2022-06-14:
Hi @sagar-k-ai,

We just notice, we put the old v2.0.4 image file at releaes 3.0.11 tag on github.
Can you try the image from here? https://tinker-board.asus.com/download-list.html?product=tinker-board-s-r2

And let us know, if any good news.

Thanks.