TinkerBoard/debian_kernel #65: please include kernel module usblp.ko in future releases

bombasticbob, 2022-06-13:
I downloaded Tinker_Board_2-Debian-Buster-v2.0.4-20211222.zip for a Tinker Board 2 and it has the 4.4.194 kernel, which I believe is in the linux4.4-rk3288-tinker_board branch.  Hopefully this is correct and I can figure out the correct configuration so that the headers match.  I need to have usblp.ko available to communicate with receipt printers.

Can you, in the future, include usblp.ko as a loadable kernel module?  This way I can just use modprobe to configure the system to load the module.  If it interferes with CUPS you can always make it not load by defaut, but it needs to at least exist in /lib to avoid having to search for the correct kernel version etc. and build it myself.

TinkerTeam, 2022-06-14:
Hi Bob,

The kernel for Tinker Board 2 is in a another repo. Following is the link with tag v2.0.4:
https://github.com/TinkerBoard2/kernel/tree/tinker_board_2-debian_10-2.0.4

Thanks.

bombasticbob, 2022-06-14:
thanks, that was very helpful (for some reason I could not locate this repo).

I did manage to succesfully build the usblp module using kernel source with a matching version (4.4.194) from this repo, grabbing the config from /proc/Config.gz, modifying the correct line in .config (I think it was USB_PRINTER=m), then using "make oldconfig" and "make modules" to build the modules, and then copying that one file to the right spot and using "depmod" and "modprobe" to install it.  The printer appears to function correctly now.  Still, it's nicer if the module is there already and I do not need matching kernel source etc. to build it.  [it has been a few years since I built all or part of the Linux kernel though]