TinkerBoard/debian_kernel #67: Tinker Board S Debian 10 V3.0.11 fails to boot on Tinker Board S

CodeInquisitor, 2022-08-03:
After downloading the Tinker Board S Debian 10 V3.0.11 image
from this page: https://tinker-board.asus.com/download-list.html?product=tinker-board-s
and flashing to SD card,
Tinker Board S was not able to boot successfully.
After powering on, screen remains blank and no text ever appears.
Using the Tinker Board Debian Stretch V2.2.9 image produces the same result, but the Tinker Board Debian OS V2.1.16 image can work normally.
Does the Tinker Board S Debian 10 V3.0.11 image not support Tinker Board S?

asusiot, 2022-09-07:
hello @CodeInquisitor, 
it does, is your board on the sd card boot mode? no jumper's plugged?

CodeInquisitor, 2022-09-08:
> hello @CodeInquisitor, it does, is your board on the sd card boot mode? no jumper's plugged?

Hello @asusiot,
No, no jumper was plugged. Is forced sd card boot mode required for booting Debian 10 V3.0.11 image? How come Debian OS V2.1.16 image does not require it?