TinkerBoard/debian_kernel #68: Audio output start hissing, noise

DIKIIDM, 2023-01-11:
After some time of continuous working (after 2 days, some time a week), the audio output from jack 3.5 starts hissing, noise (shhhhhhh).
Whether it's a youtube video in the browser, or it's my mp3 playlist playing.
Only reboot can help

I installed the Tinker Board Debian 10 V3.0.11 image on my Tinker board R2.0.
I use it as an audio player through jack 3.5.

Before R2.0 I used a regular Tinker board and did not face this problem.

Has anyone encountered a similar problem, what could be the matter?

DIKIIDM, 2023-01-17:
same issue
https://tinker-board.asus.com/forum/index.php?/topic/15304-tinker-board-r20-audio-problem-hissing-noise/

Malkijan, 2023-02-18:
I tried so many things on alsamixer's settings like increasing buffer_size and period_size and still getting the issue.
Sometimes sounds perfect goes on but whenever I restart the device problem occurs in a couple of hours.

Malkijan, 2023-02-18:
I have a Tinker Board S, that plays music without any problem but with the same configuration my other device Tinker Board R2.0 plays so badly almost every day, at least once gives white noise (hissing noise as DIKIIDM mentioned)
Has anyone found a solution?

pyavitz, 2023-03-04:
I have the same issue and there appears to be no rhyme or reason for it. Sometimes it happens right away and other times after 48 hours or so. Unfortunately I can't seem to find anything in the logs or dmesg concerning the problem. Its as if a switch gets thrown and everything that comes out is just white noise static.

Could this be power related in some way?

serkansmg, 2023-03-09:
> same issue https://tinker-board.asus.com/forum/index.php?/topic/15304-tinker-board-r20-audio-problem-hissing-noise/

check this link again.. i've posted a solution for it.. not to make it duplicate.. just check it..

justinasjaronis, 2023-05-05:
I did this on Asus Tinker Board S R2.0, and